bennymeg/Fabrication-Toolkit #161: BOM doesn't include the part that not on the board

ceezblog, 2024-04-27:
I have a small issue with the part that not on the board that not make into the BOM.

It is a header (male) and pluggable terminal block (female). As only the header is placed on the PCB, but not the pluggable terminal block. Since I want to order everything from JLC to simplify the process, I put a symbol for the pluggable terminal block as well with intention for it to make into the BOM and put a remark on order telling JLC the extra connector is just coming along with the board.

I tried all combination of DNP and Exclude from board, but not Exclude from bill of materials, none worked.
[image]

So I notice if the part on schematic doesn't have footprint or marked with "Exclude from board" then the output of BOM doesn't have it.

I have been using Kicad 7, Kicad 8 and having the same issue.

For now, manually add this part into the BOM works for me.

Thanks

troky, 2024-08-06:
I have the exact same problem.

perryc, 2024-08-10:
JLC tell me they need the footprint to be a "-"  in the bom to include a loose part with a pcb (in my case an antenna).  So how about in sch if footprint is "-" and attribute "exclude for board" is set it get added to the bom?