bennymeg/Fabrication-Toolkit #168: [Feature Request]: Folder location

robodna, 2024-08-10:
### Preflight Checklist

- [X] I have searched the [issue tracker](https://www.github.com/bennymeg/Fabrication-Toolkit/issues) for a feature request that matches the one I want to file, without success.

### Problem Description

Using Kicad8.04, the fabrication toolkit does not allow the user to choose a file path to save the production files, and does not automatically name the folders as it did with KiCad7.

### Proposed Solution

Allow user to choose a destination folder and folder name.

### Alternatives Considered

Manually re-jiggering the resulting files into their own production folder for the version of the PCB.

petaflot, 2025-01-23:
where does it even save the files? After I click "Generate" there is the progress bar, the plugin window closes and I can't find anything that look like production files and an archive of these

martiinezz, 2025-07-09:
In my case folder is in the same place as .kicad_pcb file