bennymeg/Fabrication-Toolkit #172: [Bug]: Fabrication output includes all layers

astuder, 2024-09-08:
### Preflight Checklist

- [X] I have searched the [issue tracker](https://www.github.com/bennymeg/Fabrication-Toolkit/issues) for a bug report that matches the one I want to file, without success.

### Fabrication Toolkit Version

4.4.1

### What operating system are you using?

Ubuntu

### Operating System Version

Ubuntu 24.04.1 LTS

### What arch are you using?

x64

### Expected Behavior

The exported Gerber files should only include commonly used layers (like copper, silk-screen, mask, paste, edge cuts..).

### Actual Behavior

The exported gerber ZIP file includes all layers, including adhesive, user1-9, courtyard, etc.

### Additional Information

This is on a fresh install of Kicad 8.0.4 and the Fabrication Plugin. 

fabrication-toolkit-options.json: {"EXTRA_LAYERS": "", "EXTEND_EDGE_CUT": false, "AUTO TRANSLATE": true, "AUTO FILL": true, "EXCLUDE DNP": true}

patzf, 2024-09-28:
Just wanted to chime in - just wanted to file a "bug" report for this behaviour although it does seem intended. But I strongly agree that it would be super useful to at least have some way of defining "presets" of teh layers that should be exported so only the layers of relevance go into the gerber file zip. As of now, anytime I rerun teh fab toolkit i have to manually delete the unneeded gerbes afterwards which is error prone

bennymeg, 2024-10-18:
This is actually a bug. It was not intended.
Is 4.4.1 is he first time you encounter this behavior? or did you notice it in an earlier version.

astuder, 2024-10-18:
I didn't have the issue in the past on Windows, but I'm not sure whether that was with KiCad 7. I just tested it with KiCad 8.0.5, Fabrication Toolkit 4.4.1 on Windows 11 and it has the same problem.

I also just updated to Fabrication Toolkit 4.5 on Windows 11 and it still exports all layers.

astuder, 2024-10-18:
And just tested with KiCad 7, Windows 11, using Fabrication Toolkit 3.1, and it works as expected. Opening the same project in KiCad 8, and exporting using Fabrication Toolkit 4.5 exports all layers.

bennymeg, 2024-10-20:
This is not a bug after all, this is a feature that had been added by the community.
The plugin iterates over all the active layers and add them (instead of manual configuration).
In order to ignore layers, you simply need to disable them.
This can be done by navigating to: `File > Board Setup > Board Stackup > Board Editor Layers` and by removing the check mark from the unwanted layers.

closed 2024-10-20 by bennymeg

astuder, 2024-10-20:
Your tool, your call.

But from my perspective as user, there are layers that I need when designing a board, while I don't want them exported when sending a job to JLC. For example I use the User layers for dimensions, or physical restrictions I need to design around. So removing those layers from the board is a no-go for me.

Besides the BOM and POS files, not having to fiddle with the Gerber export dialog was the main benefit of the Fabrication Toolkit plugin. 

If you want to avoid creating your own layer selection, maybe you can read the current selection from KiCad's gerber export?

astuder, 2024-10-20:
PS: And there are layers like courtyards, which can't be deselected in the board stackup dialog (not like there's a scenario where one would want to actually do that).

bennymeg, 2024-10-21:
I am always open to changes.
I do not think that using visibility is the correct approach.
Probably the best solution is to render all the layers as checkbox's and let the user decide, but currently I do not have time to implement it, maybe in future release.
For now, I think you can remove most of the unwanted layers from pcb and should be left only with small number of extra layers in the production files.

astuder, 2024-10-21:
Sorry, missed your response as I was busy writing the patch :)

I opted for a check-box to keep things simple. Together with the field for extra layers, users have flexibility to cherry pick layers on top of the standard layers.