bennymeg/Fabrication-Toolkit #187: [Bug]:

thestumbler, 2025-01-17:
### Preflight Checklist

- [x] I have searched the [issue tracker](https://www.github.com/bennymeg/Fabrication-Toolkit/issues) for a bug report that matches the one I want to file, without success.

### Fabrication Toolkit Version

5.0.0

### KiCad Version

8.0.7 release build

### What operating system are you using?

Windows

### What arch are you using?

arm64 (including Apple Silicon)

### Last Known Working Fabrication Toolkit version

n/a

### Expected Behavior

All components with a properly assigned LCSC vendor stock number should be picked up by the tool and the subsequent JLCPCB import process.

### Actual Behavior

I have one component which appears normal, but fails to be imported by JLCPCB. It is a 2-circuit 1.27mm pin header thru-hole. The LCSC stock number is [C41370734](https://www.lcsc.com/product-detail/Pin-Headers_Megastar-ZX-PZ1-27-1-2PZZ_C41370734.html?s_z=n_C41370734). 

By all appearances, this stock number is correctly in the BOM which the Fabrication Toolkit generates:
`JP1,PinHeader_1x02_P1.27mm_Vertical,1,TERM,C41370734
`

### Additional Information

The only hint I can see here is that this stock number is longer than all the other numbers in the BOM. The largest number of the "working" parts is C + 7 digits (8 chars total). This jumper's stock number is C + 9 digits (9 chars total).

thestumbler, 2025-01-21:
"Fails" may not be exactly the correct word. There are no error messages upon BOM upload. The resulting uploaded BOM was in three sections: (1) normal parts that were recognized and has sufficient inventory, (2) parts that were recognized but had insufficient inventory, and (3) parts that were not recognized. "Recognized" meaning the LCSC stocking number was found to be valid.

In the case of this 2-pin 50 mil jumper, the system didn't recognize the LCSC part number. When I manually added the part number, everything was okay. 

I will be uploading some more PCB designs today and tomorrow, and will try to make an example to demonstrate this. 

-Chris

bennymeg, 2025-01-28:
I think that happened because that part [doesn't exist on JLCBPCB parts library](https://jlcpcb.com/parts/componentSearch?searchTxt=C41370734).
I will close this for now since it not an issue with the plugin.

closed 2025-01-28 by bennymeg

thestumbler, 2025-01-29:
Thanks for checking this out, guys. There was definitely something weird happening. Part of the problem was also this was my first time attempting a PCB assembly from JLCPCB, so there was some new user unfamiliarity going on as well. 

- [ ] I ultimately learned that JLCPCB "parts" are a subset of LCSC parts, and the process of adding "LCSC but not JLCPCB" parts is completely different from adding outside distributor parts. So LCSC is not really internal, but not really external, either. The other issue was that using a not-JLCPCB part meant your project would pause until the parts were in-hand. So I just adjusted my mindset to only use JLCPCB parts.