bennymeg/Fabrication-Toolkit #189: [Bug]: JLCPCB BOM import tool chooses "Value" field over "LCSC Part #"

GrandFatherADI, 2025-01-22:
### Preflight Checklist

- [x] I have searched the [issue tracker](https://www.github.com/bennymeg/Fabrication-Toolkit/issues) for a bug report that matches the one I want to file, without success.

### Fabrication Toolkit Version

5.0.0

### KiCad Version

8.0

### What operating system are you using?

Windows

### Operating System Version

Windows 10 Pro 22H2

### What arch are you using?

x64

### Expected Behavior

JLCPCB BOM importer to choose component part numbers from the "LCSC Part #" column in the bom.csv.

### Actual Behavior

JLCPCB BOM importer chooses component part numbers from the "Value" column in the bom.csv.

### Additional Information

Altering process.py to not generate the "Value" column (comment out line 254 in process.py) fixes the issue.

Reordering the columns (move line 254 so it follows the 'LCSC Part #' line) makes no difference.

GrandFatherADI, 2025-01-26:
Feedback from JLCPCB indicates they use the "Comment" column and "Footprint" column to match components. Updating process.py to populate the Comment field in the generated bom.csv file using the contents of my MPN field gives correct results.

If there is interest I'll generate a pull request that adds the ability for users to remap footprint fields to bom.csv columns to suit their preferred convention for manufacturer part number field.

bennymeg, 2025-01-28:
> Note however that the column name should be changed from "LCSC Part #" to "JLCPCB Part #" to conform to a recent update in the import tool.

There are many options for the keys to enable support for working together with other tools and plugins that use these keys.

> Feedback from JLCPCB indicates they use the "Comment" column and "Footprint" column to match components. Updating process.py to populate the Comment field in the generated bom.csv file using the contents of my MPN field gives correct results.
If there is interest, I'll generate a pull request that adds the ability for users to remap footprint fields to bom.csv columns to suit their preferred convention for manufacturer part number field.

If you wish to submit a PR that populate a "Comment" field with the MPN in case there are both MPN (or a variant of it) and a JLC/LCSC (etc.) field I will happily merge it.