bennymeg/Fabrication-Toolkit #195: [Bug]: KiCAD v9 JLCPCB Rotation Offset attribute ignored

romain145, 2025-01-31:
### Preflight Checklist

- [x] I have searched the [issue tracker](https://www.github.com/bennymeg/Fabrication-Toolkit/issues) for a bug report that matches the one I want to file, without success.

### Fabrication Toolkit Version

5.0.1

### KiCad Version

9.0.0-rc2

### What operating system are you using?

Windows

### Operating System Version

Windows 11 Version 23H2

### What arch are you using?

x64

### Expected Behavior

On KiCAD v9 a change in the JLCPCB Rotation Offset in schematic, then imported into the PCB (yes I didn't forget that step) and correctly showing in the footprint properties does not correctly export to the positions.csv file.

[image]

[image]

Fox context, the position was previously set to 90° and I changed it from 90 to 0.
Deleting the production folder did not fix the issue.

May be related to #156 

### Actual Behavior

```Application: KiCad PCB Editor x64 on x64

Version: 9.0.0-rc2-142-g42b9a9604b, release build

Libraries:
	wxWidgets 3.2.6
	FreeType 2.13.3
	HarfBuzz 10.0.1
	FontConfig 2.15.0
	libcurl/8.10.1-DEV Schannel zlib/1.3.1

Platform: Windows 11 (build 22631), 64-bit edition, 64 bit, Little endian, wxMSW
OpenGL: ATI Technologies Inc., AMD Radeon RX 6700 XT, 4.6.0 Compatibility Profile Context 21.10.34.230823

Build Info:
	Date: Jan 26 2025 06:17:17
	wxWidgets: 3.2.6 (wchar_t,wx containers)
	Boost: 1.86.0
	OCC: 7.8.1
	Curl: 8.10.1-DEV
	ngspice: 44
	Compiler: Visual C++ 1942 without C++ ABI
	KICAD_IPC_API=ON

Locale: 
	Lang: en_GB
	Enc: UTF-8
	Num: 1,234.5
	Encoded кΩ丈: D0BACEA9E4B888 (sys), D0BACEA9E4B888 (utf8)
```

romain145, 2025-01-31:
Also, yes I did try changing the field name from 'JLCPCB Rotation Offset' to 'FT Rotation Offset' as per the latest instructions on the Readme:

[image]

> changing the field name isn't great and will break working projects and is even more confusing if the screenshot shows the old name </rant mode off>

The J8 property is correct in the .kicad_pcb file:

```
		(property "FT Rotation Offset" "0"
			(at 0 0 90)
			(unlocked yes)
			(layer "B.Fab")
			(hide yes)
			(uuid "52dfa3fe-ba90-4c21-a3d5-585d91f5d0bc")
			(effects
				(font
					(size 1 1)
					(thickness 0.15)
				)
				(justify mirror)
			)
		)
```

Am I missing something?

NieLuc, 2025-02-24:
While some of my symptoms are the same, for me it doesnt even add any components that I added while using KiCad 9.0. Also keeps the old positions when I move the footprints

bennymeg, 2025-02-27:
did you remembered to sync the PCB after you fixed the attribute name?

bennymeg, 2025-02-27:
You wrote you did it with the old attribute but not with the new attribute, so I wanted to verify.
I don't see a reason why this should fail on v9, I will have to check.
Anyway, v9 is not officially supported by this plugin yet.

funvill, 2025-03-04:
It looks like I got hit by this issue as well. 

[image]

https://bsky.app/profile/did:plc:swvbm2bk3irkyzp3jbeoymsc/post/3ljjm7sv4is2x

My fault for not checking for support before upgrading to KiCad9. 

KiCad Version: 9.0.0, release build

Fabrication Toolkit Version: 5.0.1

cyberhuman, 2025-05-04:
Has anyone found a solution for this?
It's quite cumbersome to fix rotation by hand in the JLCPCB online editor.

ImogenWren, 2025-07-31:
As almost all my exported components are rotated 180deg, wouldnt it be easier to offer a manual override for all components to reference them with a 180deg offset, rather than relying on modifying every single part we have used?