bennymeg/Fabrication-Toolkit #199: [Feature Request]: support for multiple PCBs in single file

petaflot, 2025-03-02:
### Preflight Checklist

- [x] I have searched the [issue tracker](https://www.github.com/bennymeg/Fabrication-Toolkit/issues) for a feature request that matches the one I want to file, without success.

### Problem Description

# Problem

some complex PCB setups require multiple yet dependent boards, either because they need to be stacked or assembled like a box. these PCBs are often part of a single schematic (for practical and readability reasons) so it is not convenient to have separate projects for each sub-board. also, having everything in the same file makes things like alignment much easier

## Manufacturing aspect

from what I gathered, PCB manufacturers do not accept multiple board outlines in a single file. from a machinist point of view, multiple boards in a file makes things like picking the boards a nightmare. there is little power on our side to change this.

# Workaround

a workaround I've found is to have all the sub-PCBs in a single file ; prior to exporting, selectively delete all items (components and board outlines) that do not belong to a chosen part. export, undo, selectively delete all items that do not belong to another sub-PCB, export, and repeat as many times as required.

## Problem

this is time consuming and prone to errors

### Proposed Solution

it seems a cleaner solution would be to allow a user to create named zones in a chosen layer (ie. User 7) ; upon export, fabrication toolkit should sequentially export the content of each zone (ignoring any items that are outside of this zone) and use the name of the zone as a suffix, so if I have a file called `cube.kicad_pcb` each of the 6 individual faces should be automatically exported to `cube_bottom.zip`, `cube_top.zip`, `cube_front.zip`, `cube_back.zip`, `cube_left.zip` and `cube_right.zip`

if zones are rectangular, it is very easy to compute if something is inside or outside of a zone. extra care must be taken when footprints overlap a zone's boundary.

BOM and CPL files should also be generated accordingly.

### Alternatives Considered

* manual labor (as described above)
* different kicad_pcb files

petaflot, 2025-03-02:
started to work on it here: https://github.com/petaflot/Fabrication-Toolkit/tree/splitpcb

help welcome! (typically for line 54 in thread.py: "# TODO loop over all rule areas in layer 7 that have no options checked"

bennymeg, 2025-03-05:
Thank you for the elaborated request,
I am a bit busy at the moment so I could not develop something like that at the moment.

I guess you can do something along this way:
Target v9+ and try to utilize the new component class feature.
Then add a dropdown menu with all the component classes and "all" so the user can select which class to export.
Finally, you can filter all the components by class at the plugin BOM and CPL generation functions accordingly.