bennymeg/Fabrication-Toolkit #200: [Bug]: Regression: "'FOOTPRINT' object has no attribute 'HasProperty'"

Erhannis, 2025-03-16:
### Preflight Checklist

- [x] I have searched the [issue tracker](https://www.github.com/bennymeg/Fabrication-Toolkit/issues) for a bug report that matches the one I want to file, without success.

### Fabrication Toolkit Version

5.0.1

### KiCad Version

9.99 (nightly)

### What operating system are you using?

Ubuntu

### Operating System Version

Ubuntu 24.10

### What arch are you using?

x64

### Last Known Working Fabrication Toolkit version

8, I think?

### Expected Behavior

When I click "Fabrication Toolkit" and complete the dialog box, it should create fabrication files.

### Actual Behavior

It starts to process, then shows error "'FOOTPRINT' object has no attribute 'HasProperty'".

### Additional Information

I see this was once a problem ( #130 and a bunch of others), but I'm seeing it now with kicad 9 nightly.  I don't know why it would have come back.  I'm working on a project I created last year, is it possible something weird carried over?  I think Fabrication Toolkit worked on the same project with kicad 8.

Erhannis, 2025-03-17:
So, two things:
1. I think the version selection code has an off-by-one error?  It's currently
```python
def is_v9(version = get_version()):
    return version >= 8.99 and version < 9.99
```
but notably that means that `8.99` is version 9, and `9.99` is not version 9.  I imagine it should instead be
```python
def is_v9(version = get_version()):
    return version > 8.99 and version <= 9.99
```
and same with the other four functions, unless you're trying to make some statement about nightly not being the same as release.
2. In kicad commit c5bb59468d733ddde8c663ea27d89e190723b70d , `HasFieldByName` and `GetFieldByName` respectively were renamed to `HasField` and `GetField`.

Erhannis, 2025-03-17:
Addendum: considering that trying to fix 2 would mean trying to figure out how to support both the current api and the changes in nightly, maybe that's why the nightly versions weren't considered the same as their release versions?  In which case, carry on.

bennymeg, 2025-03-18:
Correct, nightly build contains unstable breaking changes, so that is the reason for this selected methodology and behavior.
I'll add a fix that you will be able to test from the repo shortly.

closed 2025-03-20 by bennymeg