bennymeg/Fabrication-Toolkit #204: [Question] Kicad Jobset command

EFeru, 2025-04-15:
Hi @bennymeg,

Thanks for this awesome tool. As you might know, Kicad 9.0 has introduced Jobset to automate out workflow:
https://docs.kicad.org/9.0/en/kicad/kicad.html#jobsets

What would be the command to trigger this plugin? What I have tried below does not seem to work:
[image]

Thanks for your help.

bennymeg, 2025-04-15:
I have not got a chance to play with v9 yet.
I assume the solution would be using the plugin cli function to generate the production files.
Try using the [KICAD9_3RD_PARTY](https://docs.kicad.org/9.0/en/kicad/kicad.html#kicad-environment-variables) variable to invoke the plugin. 
Probably something like `python3 -m ${KICAD9_3RD_PARTY}\plugins\com_github_bennymeg_JLC-Plugin-for-KiCad\cli.py -p /myProject/myBoard.kicad_pcb`

EFeru, 2025-04-15:
Thank you for your quick answer. Unfortunately it does not work. I try also running the command below in a command prompt window which gave the following error. I am not sure what am I missing because in the Kicad GUI the Toolkit works flawesly.
```
C:\Users\eferu\Documents\KiCad\9.0\3rdparty\plugins>python -m com_github_bennymeg_JLC-Plugin-for-KiCad.cli -h
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\eferu\Documents\KiCad\9.0\3rdparty\plugins\com_github_bennymeg_JLC-Plugin-for-KiCad\cli.py", line 3, in <module>
    from .thread import ProcessThread
  File "C:\Users\eferu\Documents\KiCad\9.0\3rdparty\plugins\com_github_bennymeg_JLC-Plugin-for-KiCad\thread.py", line 2, in <module>
    import wx
ModuleNotFoundError: No module named 'wx'
```
Only `python` is recognized on my Windows machine, `python3` not even though I have [Python 3.13.3](https://www.python.org/downloads/release/python-3133/) installed.

EFeru, 2025-05-03:
`pip3 install wx` was not recognized in Windows, so I installed `pip install wxPython`

After that I get:
```
C:\Users\eferu\Documents\KiCad\9.0\3rdparty\plugins>python -m com_github_bennymeg_JLC-Plugin-for-KiCad.cli -h
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\eferu\Documents\KiCad\9.0\3rdparty\plugins\com_github_bennymeg_JLC-Plugin-for-KiCad\cli.py", line 3, in <module>
    from .thread import ProcessThread
  File "C:\Users\eferu\Documents\KiCad\9.0\3rdparty\plugins\com_github_bennymeg_JLC-Plugin-for-KiCad\thread.py", line 3, in <module>
    import pcbnew  # type: ignore
    ^^^^^^^^^^^^^
ModuleNotFoundError: No module named 'pcbnew'
```

ChatGPT says:
"
pcbnew is not an installable package on PyPI — it's a built-in module provided by KiCad for scripting and plugins, especially in KiCad’s own Python environment.
To use pcbnew in your scripts, you must use the Python interpreter that comes with KiCad. This environment has pcbnew and other internal modules preloaded.
"

I propose to put this on hold for now, and when you have time you can give a try to https://docs.kicad.org/9.0/en/kicad/kicad.html#jobsets 
The advantage of jobset wiht the Frabrication-Toolkit would be that I can create a release with only one click (otherwise it will require some manual folder management).

bennymeg, 2025-06-09:
I assume this will only work with KiCad's distributed version of python only.
It would take some time until I will upgrade, I'm in the middle of couple of projects currently.
Could you try: `C:\Program Files\KiCad\9.0\bin\python -m ${KICAD9_3RD_PARTY}\plugins\com_github_bennymeg_JLC-Plugin-for-KiCad\cli.py -p /myProject/myBoard.kicad_pcb`
You can define `C:\Program Files\KiCad\9.0\bin` as KICAD9_BIN_DIR in KiCad's paths to improve readability later if that works.
let me know how its go's.

neilenns, 2025-10-15:
I spent a bunch more time mucking about with this, now on Windows, and am no closer to getting it working :(

I was able to install `wx` by doing the following:

```powershell
& "C:\Program Files\KiCad\9.0\bin\python.exe" -m pip install wxPython
```

However I can't find any combination of command line and environment variable that will successfully run the script as a module.

If I set `PYTHONPATH` in my environment then my python install will run the script fine but fails because there's no module named pcb new:

```powershell
python -m com_github_bennymeg_JLC-Plugin-for-KiCad.cli
```

If I try and run it using the KiCad python I can't get past the problem of it not being treated as a module:

```powershell
 $env:PYTHONPATH = "D:\OneDrive\Documents\KiCad\9.0\3rdparty\plugins"
& "C:\Program Files\KiCad\9.0\bin\python.exe" -m com_github_bennymeg_JLC-Plugin-for-KiCad.cli
C:\Program Files\KiCad\9.0\bin\python.exe: Error while finding module specification for 'com_github_bennymeg_JLC-Plugin-for-KiCad.cli' (ModuleNotFoundError: No module named 'com_github_bennymeg_JLC-Plugin-for-KiCad')
```

Using the command suggested back on June 9th inside the jobset configuration doesn't work either:

```
C:\Program Files\KiCad\9.0\bin\python -m D:\OneDrive\Documents\KiCad\9.0\3rdparty\plugins\com_github_bennymeg_JLC-Plugin-for-KiCad\cli -p myBoard.kicad_pcb

C:\Program Files\KiCad\9.0\bin\python.exe: Error while finding module specification for 'D:\\OneDrive\\Documents\\KiCad\\9.0\\3rdparty\\\\plugins\\com_github_bennymeg_JLC-Plugin-for-KiCad\\cli.py' (ModuleNotFoundError: No module named 'D:\\OneDrive\\Documents\\KiCad\\9'). Try using 'D:\OneDrive\Documents\KiCad\9.0\3rdparty\\plugins\com_github_bennymeg_JLC-Plugin-for-KiCad\cli' instead of 'D:\OneDrive\Documents\KiCad\9.0\3rdparty\\plugins\com_github_bennymeg_JLC-Plugin-for-KiCad\cli.py' as the module name.
```

(using `cli` instead of `cli.py` doesn't work either)

On Linux I can run the script fine via the jobset, but `pcbnew.LoadBoard()` always returns `None`.

radoslav06, 2025-11-18:
I'm no expert, but maybe try to run as a normal script (not a module)? I currently don't have this plugin installed, but I use the following command for the [iBom ](https://github.com/openscopeproject/InteractiveHtmlBom) plugin, you can adopt:

```bash
python "${KICAD9_3RD_PARTY}\plugins\org_openscopeproject_InteractiveHtmlBom\generate_interactive_bom.py" "${PROJECTNAME}.kicad_pcb" --extra-fields "ManufacturerPN,LCSC,kicad_dnp" --extra-data-file "${PROJECTNAME}.kicad_pcb" --no-browser --name-format "${PROJECTNAME}" --dest-dir ${JOBSET_OUTPUT_WORK_PATH}
```