bennymeg/Fabrication-Toolkit #216: [Bug]: KiCad 9.0.4 and Toolkit 5.1.0 no icon

tcurdt, 2025-09-13:
### Preflight Checklist

- [x] I have searched the [issue tracker](https://www.github.com/bennymeg/Fabrication-Toolkit/issues) for a bug report that matches the one I want to file, without success.

### Fabrication Toolkit Version

5.1.0

### KiCad Version

9.0.4

### What operating system are you using?

macOS

### Operating System Version

15.5 (24F74)

### What arch are you using?

arm64 (including Apple Silicon)

### Last Known Working Fabrication Toolkit version

not sure - many

### Expected Behavior

See an icon to trigger the export

### Actual Behavior

I do not see an icon anymore.

[image]

### Additional Information

Seems like this has been an issue before.

bennymeg, 2025-09-14:
Please verify that the plugin is installed.
KiCad does not maintain plugins on update.

tcurdt, 2025-09-14:
[image]

seuchato, 2025-09-14:
> Please verify that the plugin is installed. KiCad does not maintain plugins on update.

Well, I am an Engineer. I do not like to waste anybody's time. I see the exact same output as tcurdt has posted a few hours ago (So 14 Sep 2025 20:xx:yy CEST)

What else is required? Kindly let me know.
Greez
Chris

seuchato, 2025-09-17:
I Just set up a virtualbox Opensuse Tumbleweed VM.
KICAD Version (from standard Tumbleweed repo) is 9.04 and Fabrication Toolkit 5.1.0. 
I see the icon and the Toolkit seems to work, at least is processing...
@tcurdt : maybe that's a way to get around the problem? I don't know about VirtualBox on Mac though, never had one.
greez
chris

closed 2025-09-18 by tcurdt